sea-bass/pyrobosim #2: [Planning] Integrate PDDLStream task and motion planner

sea-bass, 2022-05-08:
A big reason for this repo existing is that it can be used to prototype task and motion planners.

One such planner which will be used as the main demo is PDDLStream (https://github.com/caelan/pddlstream).

We should integrate a few demos that use this planner as a regular PDDL planner (with only discrete variables), plus with the addition of continuous variables such as sampling placement poses and navigation plans (e.g., RRTs).

We will also have to figure out the necessary steps / instructions / scrips to ensure that PDDLStream works easily with this package.

roshandhakal, 2022-05-31:
I am excited for this. I am working on PDDLStream for my project as well.

sea-bass, 2022-06-26:
Closed by #14 and #17.

closed 2022-06-26 by sea-bass