sea-bass/pyrobosim #15: [Planning] RRT Implementation Improvements

sea-bass, 2022-06-12:
I found 2 annoying things about the RRT implementation in testing other stuff:

We don't check if we can immediately connect to the goal over short distances. So some trees will look silly like this:

[image]

Furthermore, in a bidirectional RRT* implementation we don't try rewire the tree once we connect the start and end nodes. You can see that in the kink here, for example:

[image]

This second one might be a little trickier to do in that in may actually require the solution path to be added to the start tree and then running a rewire on all the newly connected nodes.

ibrahiminfinite, 2023-01-13:
Are there any specific examples of where this happened?

ibrahiminfinite, 2023-01-13:
I am able to replicate the first issue 

    start = Pose(x=-2.2, y=2.8)
    goal = Pose(x=-2.2, y=3.0)

[image]

sea-bass, 2023-01-13:
I suspect this is a combination of using bidirectional RRT and RRT* for path shortening, in that the RRT* step isn't shortening the connection between start and goal trees.

closed 2023-04-23 by sea-bass