sea-bass/pyrobosim #20: Demo script: No module named `pddlstream`

AndyZe, 2022-08-09:
I believe I've followed the instructions here to the letter but get this error:

```
[andy-G7-7500][main] pyrobosim$ python3 examples/demo_pddl.py --example 01_simple verbose
Traceback (most recent call last):
  File "/home/andy/ws_tp/pyrobosim/pyrobosim/examples/demo_pddl.py", line 13, in <module>
    from pyrobosim.planning.pddlstream.planner import PDDLStreamPlanner
  File "/home/andy/.local/lib/python3.10/site-packages/pyrobosim/planning/pddlstream/planner.py", line 4, in <module>
    from pddlstream.algorithms.focused import solve_focused
ModuleNotFoundError: No module named 'pddlstream'
```

I've definitely already run this:  `./setup/setup_pddlstream.bash`

Cool blog and package btw!

ibrahiminfinite, 2022-08-09:
Hi @AndyZe 

Just wanted to clarify if you ran the pddlstream setup script before or after your sourced your virtual env.
If it is before you activated the env ,  pyrobosim would not be able to find it as pddlstream would get installed outside the env.

AndyZe, 2022-08-09:
Yep, that was it. Thanks.