sea-bass/pyrobosim #23: Why do some predicates have `get()`, some do not?

AndyZe, 2022-09-10:
A question I ran across as a I started to work with this. From the `04_nav_manip_stream` demo, why do some of the parameters in the goal specification include `get()`, others don't?

>     goal_literals = [
>         ("Has", get("desk0_desktop"), get("banana0")),
>         ("Has", "counter", get("apple1")),
>         ("HasNone", get("bathroom"), "banana"),
>         ("HasAll", "table", "water"),
>     ]

counter, banana, table, water do not have `get()`

I'm guessing it is something about the problem setup that there can be multiple robots, rooms, apples, and bananas.

sea-bass, 2022-09-10:
Admittedly, this is something where I got a bit lazy and sacrificed user-friendliness, so it's great that you've officially made an issue for it.

Your question gets it right. Basically, there are categories (e.g. `banana`, `desk`) which are handled as regular strings, and then instances (e.g., `banana0`, `my_desk`) which are handled as the actual Python objects.

So to go from the string `banana0` to the actual Python object, the examples use `get()` as a convenience lambda.

Of course, nothing stops the planner from parsing the goal specification and doing that conversion internally so users can simply enter the string `"banana0"` and we apply some simple logic to determine whether this is a category or object name.

Will take this on at some point!

sea-bass, 2022-09-15:
Thanks again for prompting on this -- should be addressed by https://github.com/sea-bass/pyrobosim/pull/25.

closed 2022-09-16 by sea-bass