sea-bass/pyrobosim #24: What is a static vs fluent predicate?

AndyZe, 2022-09-10:
Looking at the blog here:  https://roboticseabass.com/2022/07/30/integrated-task-and-motion-planning-in-robotics/

It would be nice if there were an explanation of `static` vs `fluent`.

sea-bass, 2022-09-10:
Good question. Is this something you wanted to see in the blog? In the repo? Both?

Static is something immutable that is typically generated with the initial condition. For example `(Apple apple0)`, or "apple0 is of type Apple, and will forever be the case".

Fluent, on the other hand, is something that can be changed by executing an action. For example, `(At apple0 desk0`) which can be applied by a `place` action or removed by a `pick` action in our formulation.

AndyZe, 2022-09-12:
Thanks!

> 
Is this something you wanted to see in the blog? In the repo? Both?

Maybe https://pyrobosim.readthedocs.io/en/latest/ would be the best place. I guess the blog is disconnected from this repo - a lot of people might miss the blog.

closed 2022-09-12 by AndyZe