sea-bass/pyrobosim #29: Clicking on Navigate raises a ValueError

DigitalGabriele, 2022-09-28:
Hi,

I followed your documentation and ran the following line:

`ros2 run pyrobosim_ros demo.py`

I saw a nice map. I clicked on Navigate and received the following error:

`~/pyrobosim/pyrobosim$ ros2 run pyrobosim_ros demo.py
[INFO] [1664361758.672906297] [pyrobosim]: World node started.
[INFO] [1664361758.673251481] [pyrobosim]: Creating demo world programmatically.
Navigating to desk0_desktop
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/home/gabriele-plankton/pyrobosim/install/pyrobosim_ros/lib/python3.8/site-packages/pyrobosim/navigation/execution.py", line 56, in execute
    (traj_t, traj_x, traj_y, traj_yaw) = interpolate_trajectory(traj, self.dt)
  File "/home/gabriele-plankton/pyrobosim/install/pyrobosim_ros/lib/python3.8/site-packages/pyrobosim/navigation/trajectory.py", line 97, in interpolate_trajectory
    [slerp(t).as_euler("xyz", degrees=False)[2] for t in t_interp]
  File "/home/gabriele-plankton/pyrobosim/install/pyrobosim_ros/lib/python3.8/site-packages/pyrobosim/navigation/trajectory.py", line 97, in <listcomp>
    [slerp(t).as_euler("xyz", degrees=False)[2] for t in t_interp]
  File "/usr/lib/python3/dist-packages/scipy/spatial/transform/rotation.py", line 1757, in __call__
    raise ValueError("Expected times to be specified in a 1 "
ValueError: Expected times to be specified in a 1 dimensional array, got 0 dimensions.
^CTraceback (most recent call last):
  File "/home/gabriele-plankton/pyrobosim/install/pyrobosim_ros/lib/python3.8/site-packages/pyrobosim/gui/main.py", line 177, in on_navigate_click
    self.canvas.navigate(loc)
  File "/home/gabriele-plankton/pyrobosim/install/pyrobosim_ros/lib/python3.8/site-packages/pyrobosim/gui/world_canvas.py", line 300, in navigate
    time.sleep(sleep_time)
KeyboardInterrupt
`

sea-bass, 2022-09-28:
Thanks for reporting this!

Are you on the latest `main` branch? I just modified this exact same `demo.py` to randomize the robot pose and plan a path to a random location in an infinite loop... and I don't see this error.

Does this happen to you consistently or was it that one time? Would appreciate more details so I can dig in.

DigitalGabriele, 2022-09-28:
Yes, I cloned the main branch (Ive done it maybe 30 min ago). The error keeps appearing every time. I tried also running the following command line in a separate terminal:

`ros2 run pyrobosim_ros demo_commands.py` 

I got the same error again in the `ros2 run pyrobosim_ros demo.py` terminal.

sea-bass, 2022-09-28:
Hmm, well the only other thought I had is you're on Python 3.8 while I developed on 3.10... So maybe the scipy version was different.

But I just started up a Docker container with Python 3.8 and everything seems fine on my end there too.

Just to make sure, what is your scipy version if you open up Python and enter the following?

```
>>> import scipy
>>> scipy.__version__
1.8.1.
```

Maybe you're on an older version of scipy which expects things in different format? Because even when I try pass in a 0-dimension time, even the error is different.

If that's the case, could you try a `pip3 install --upgrade scipy` on your system?

Or if you don't want to mess with your system, you could follow the recommended instructions in https://pyrobosim.readthedocs.io/en/latest/setup.html and make a virtual environment for pyrobosim:

```
cd /path/to/pyrobosim
setup/create_python_env.bash
```

DigitalGabriele, 2022-09-28:
That was it. It was an older version of scipy (1.3.x). Thanks!

sea-bass, 2022-09-28:
Amazing, glad we got it solved! Enjoy pyrobosim, and let me know if anything else :)