sea-bass/pyrobosim #56: Add a "common.pddl" domain file to reduce duplication

AndyZe, 2023-01-09:
It looks like navigate/pick/place are duplicated in all of the domain-defining pddl files. Prob could move those to a common data file.

https://github.com/sea-bass/pyrobosim/tree/main/pyrobosim/pyrobosim/data/pddlstream/domains

sea-bass, 2023-01-09:
This is a really good idea. Assuming it's possible to "include" PDDL files in others, this would save a lot of copy-pasting.

sea-bass, 2023-01-13:
At a glance, PDDLStream's `read()` function for domains is literally a thin wrapper around reading the raw file... so I don't think something like `extends` will work out of the box.

This may not be possible without a bunch of work.

AndyZe, 2023-01-13:
I mean, it can't be that hard to copy/paste some `.pddl` files together with Python then save them to a temporary datafile for reading

sea-bass, 2023-01-13:
Well... it's a little more complicated than that, sadly.

You'd have to also reorder the contents of that temp file to make sure that all sections get propertly merged, e.g., predicates from multiple files get placed under one `(:predicates )` section, etc.

And then there's the path issue. Right now, I'm assuming we have one folder with a `domain.pddl` and a `stream.pddl` file. So where should `common.pddl` go so that it's read correctly?

We would also have to ensure we read the right file, because the file name means nothing. We have to open up that PDDL file and look for e.g. `(define (domain common) ... )` to verify that is the file we want to exist.

But maybe if we grab a Python package that does all the LISP interpreting like https://pypi.org/project/pddlpy/, it may not be so bad?

closed 2023-07-16 by sea-bass