sea-bass/pyrobosim #64: Some while loops do not check for sigint or do not have sleeps

AndyZe, 2023-01-29:
Partially fixed by #63 but there are still some other while loops which I believe would hang forever when Ctrl+C is pressed. Here's an example:

pyrobosim/gui/world_canvas.py
```
        while True:
            # Check if any robot is currently navigating.
            nav_status = [robot.executing_nav for robot in self.world.robots]
            ...
            time.sleep(sleep_time)
```

Here's a simple solution:  https://stackoverflow.com/a/18994932

sea-bass, 2023-02-04:
I opened #65 to allow the GUI to close properly. The loop you pointed to in the issue description actually only runs on a separate thread, so it's not the one causing blocking.

The other loops with `while True` from a quick search also seem fine:
* One is in the RRT Implementation, which I'm not worried about
* The others are coupled with `yield` commands, so generators don't have this issue as their output is requested one by one. `while True` here simply implies that the generator won't stop yielding samples unless something inside the loop breaks it.

ibrahiminfinite, 2023-05-19:
Can this issue be closed now ?

sea-bass, 2023-05-24:
I think so.

closed 2023-05-24 by sea-bass