sea-bass/pyrobosim #79: Version pyrobosim and release to pip

sea-bass, 2023-03-17:
Having releases of `pyrobosim` that can be `pip install`ed would be amazing.

ibrahiminfinite, 2023-03-19:
One thing I am concerned about is if the pip should install PDDLStream
The reason being , most people might just require the world simulation + path planning capability with their own behaviour logic on top of it .

This is based only on my assumption about how people may want to use it.
What are you thoughts on this @sea-bass .?

sea-bass, 2023-03-19:
Unfortunately, PDDLStream isn't pip installable, so that won't work unless users install it on their own and add it to the python path.

ibrahiminfinite, 2023-03-19:
I thiat case PDDLStream tests should be skipped with a message in pytest using dependency check in case PDDLStream is not installed.
People probably wouldn't want to see failed tests due to a dependency that they don't need to use.

closed 2023-07-16 by sea-bass