sea-bass/pyrobosim #84: Add contributor guide

sea-bass, 2023-03-19:
This will be nice to have generally, and is also a requirement to submitting to [JOSS](https://joss.theoj.org/) (... you know, if I ever feel inspired)

ibrahiminfinite, 2023-04-05:
I think we should put a list of expected behaviours and  how to check  them using the demos.
Not sure if it should go in this guide or somewhere else in the documentation.

sea-bass, 2023-04-05:
> I think we should put a list of expected behaviours and how to check them using the demos. Not sure if it should go in this guide or somewhere else in the documentation.

I think tests for all these functions are what is missing. And also integration tests -- not necessarily with the GUI, but with something at the same level of abstraction.

Definitely not something for this PR.

PR #92 (2023-05-15): Add contributor guide
Closes #84.