sea-bass/pyrobosim #85: Improve automated tests to use docker images.

ibrahiminfinite, 2023-03-19:
The current automated tests take a lot of time to run.
This could be improved by using docker images which are built weekly (as suggested by @sea-bass ) and also adding a manually triggered workflow to rebuild a docker image in case of a requirement to rebuild between the automated build interval.

sea-bass, 2023-04-16:
Another option would be to use the [GitHub Actions cache](https://docs.docker.com/build/cache/backends/gha/), which is for sure going to help with the Docker tests... but not with the standalone Python ones, necessarily. Although it should be possible to use a ROS free Docker container for that too.

ibrahiminfinite, 2023-04-19:
>  Although it should be possible to use a ROS free Docker container for that too.

This will be good. The tests take too long to run right now and most of the time is spent in setting up stuff, especially the PDDLStream

sea-bass, 2024-02-07:
Seems @eholum is looking into this

closed 2024-02-07 by sea-bass