sea-bass/pyrobosim #86: Improved  test coverage

ibrahiminfinite, 2023-03-21:
Current tests only cover a small portion of the functionalities.
More tests need to be added and also should set it up such that  tests with uninstalled dependency are skipped.

A list of tests to be added will be update here once #82 is completed.

ibrahiminfinite, 2023-04-27:
From the issues faced in #82, I think [these](https://github.com/sea-bass/pyrobosim/blob/main/pyrobosim/pyrobosim/utils/knowledge.py) utils are good candidates for some unit tests.

@sea-bass  , what are some other areas that you think should have test coverage ?

sea-bass, 2023-04-27:
> @sea-bass , what are some other areas that you think should have test coverage ?

All of them. If you'd like, pick one and make tests for it :)