sea-bass/pyrobosim #104: Restore functionality to use A* on a fully-connected world roadmap

sea-bass, 2023-05-14:
I just noticed this, but https://github.com/sea-bass/pyrobosim/pull/82 actually removed some old functionality in `pyrobosim` that is no longer available.

This has to do with having a fully-connected roadmap graph with just the nodes included in the rooms / locations and directly doing A* search along this roadmap:

[image]

@ibrahiminfinite Could you confirm this is the case? Do you think you'd be able to restore this functionality by adding another planner that uses this directly? Right now, we only have A*

sea-bass, 2023-05-14:
Basically, if in the YAML file you add a robot with an `a_star` path planner that doesn't have an occupancy grid, there is an error message. I don't think this path was ever tested and doesn't seem to work.

ibrahiminfinite, 2023-05-14:
I could add a new  `WorldGraphplanner`  (name subject to change)
That builds a graph from the world entities and plans on it , 
So instead of putting  "astar" in the config we can specify this new planner.
Ideally the name should be something that implies it works on a static graph of the world.

sea-bass, 2023-05-14:
WorldGraphPlanner, or VisibilityGraphPlanner, should be good!

If that's the case, then AStarGraph should be removed and there should be an error when no occupancy grid is passed into the AStar planner?

ibrahiminfinite, 2023-05-15:
The `AStarGraph` is internally [used](https://github.com/sea-bass/pyrobosim/blob/3786cb35ac982ab533ca29ea734b2c1714001033/pyrobosim/pyrobosim/navigation/prm.py#L38) in PRM. 
Things are bit weird with the graph A* since we are using the `astar` package from pip and it does not really need a graph object or any other planner arguments, just the start and goal.

My ideal implementation would have a graph passed to the planner (similar to the grid) and would plan on the graph.
Not sure if the `astar` package can be made to work like that.

ibrahiminfinite, 2023-05-18:
@sea-bass 
Gave this some more thought.
I think the best way to solve this would be to add a graph attribute back to the `World` class and build it similar to how it was done earlier.
And then for planning, the robot position would be added to the `World.graph` before search and removed after the search .
Since the Graph A* implementation only needs the `start` and `goal` nodes.
Similar to how it is done in the [current PRM implementation](https://github.com/sea-bass/pyrobosim/blob/3786cb35ac982ab533ca29ea734b2c1714001033/pyrobosim/pyrobosim/navigation/prm.py#L69)

sea-bass, 2023-05-18:
That sounds right to me -- thanks!

ibrahiminfinite, 2023-05-18:
However, I think the behavior of the current A* is not very intuitive.
The A* graph implementation should take in and depend on a `Graph` object and the neighbors should be found using the `Graph` object rather than the `Node` themselves.

eg : 

``` python
Graph.get_neighbors(node)
```
instead of 
``` python
node.get_neighbors()
```
This change can be done, but out `Graph` implementation also needs to be modified to support this.

Should probably be the topic of separate future PR.

PR #106 (2023-05-24): Restore world graph planning capability
I went ahead and did this because I am showing the tool tomorrow and wanted it to work.

[image]

Python:
```python
planner_config = {
    "world": world,
    "max_connection_dist": 10.0,
    "compress_path": False,
}
planner = PathPlanner("world_graph", **planner_config)
```

Yaml:
```yaml
robots:
  - name: robot
    path_planner:
      type: world_graph
      max_connection_dist: 10.0
      compress_path: false
```

Closes #104
closed 2023-05-24 by sea-bass