sea-bass/pyrobosim #116: Taskplan is None

yassineyakercap, 2023-06-28:
Hello,

I'm struggling to run a Task and Motion planning with PDDL and can't seems to find a solution. I guess it's me not doing the right thing but I'm not sure what.

Here is the error I get :

```
yyaker@yyaker-virtual-machine:~/pyroborsim_ws$ ros2 run pyrobosim_ros entrepot_pddl_planner.py
[INFO] [1687934610.340197848] [entrepot_pddlstream_planner]: Planning node ready.
[INFO] [1687934610.340669487] [entrepot_pddlstream_planner]: Waiting for goal specification...
Received new goal specification!
[INFO] [1687934664.712488846] [entrepot_pddlstream_planner]: Requesting world state...
[INFO] [1687934664.738504206] [entrepot_pddlstream_planner]: Planning...

Iteration: 1 | Complexity: 0 | Skeletons: 0 | Skeleton Queue: 0 | Disabled: 0 | Evaluations: 58 | Eager Calls: 30 | Cost: inf | Search Time: 0.005 | Sample Time: 0.001 | Total Time: 0.007
Attempt: 1 | Results: 0 | Depth: 0 | Success: True | Time: 0.108
Stream plan (0, 0, 0.000): []
Action plan (0, 0.000): []
Summary: {complexity: 0, cost: 0.000, evaluations: 58, iterations: 1, length: 2, run_time: 0.116, sample_time: 0.001, search_time: 0.114, skeletons: 1, solutions: 1, solved: True, timeout: False}

Iteration: 1 | Complexity: 0 | Skeletons: 0 | Skeleton Queue: 0 | Disabled: 0 | Evaluations: 58 | Eager Calls: 30 | Cost: inf | Search Time: 0.006 | Sample Time: 0.002 | Total Time: 0.008
Attempt: 1 | Results: 0 | Depth: 0 | Success: True | Time: 0.063
Stream plan (0, 0, 0.000): []
Action plan (0, 0.000): []
Summary: {complexity: 0, cost: 0.000, evaluations: 58, iterations: 1, length: 2, run_time: 0.071, sample_time: 0.002, search_time: 0.070, skeletons: 1, solutions: 1, solved: True, timeout: False}

Iteration: 1 | Complexity: 0 | Skeletons: 0 | Skeleton Queue: 0 | Disabled: 0 | Evaluations: 58 | Eager Calls: 30 | Cost: inf | Search Time: 0.006 | Sample Time: 0.002 | Total Time: 0.008
Attempt: 1 | Results: 0 | Depth: 0 | Success: True | Time: 0.033
Stream plan (0, 0, 0.000): []
Action plan (0, 0.000): []
Summary: {complexity: 0, cost: 0.000, evaluations: 58, iterations: 1, length: 2, run_time: 0.041, sample_time: 0.002, search_time: 0.040, skeletons: 1, solutions: 1, solved: True, timeout: False}
[INFO] [1687934664.999101148] [entrepot_pddlstream_planner]: None
[INFO] [1687934664.999636181] [entrepot_pddlstream_planner]: Robot: lrobot
[INFO] [1687934665.000054886] [entrepot_pddlstream_planner]: [('At', Robot: lrobot, Room: warehouse), ('At', Object: box_A0, Object spawn: table0_tabletop)]
Traceback (most recent call last):
  File "/home/yyaker/pyroborsim_ws/install/pyrobosim_ros/lib/pyrobosim_ros/entrepot_pddl_planner.py", line 174, in <module>
    main()
  File "/home/yyaker/pyroborsim_ws/install/pyrobosim_ros/lib/pyrobosim_ros/entrepot_pddl_planner.py", line 167, in main
    rclpy.spin(planner_node)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/__init__.py", line 222, in spin
    executor.spin_once()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 713, in spin_once
    raise handler.exception()
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/task.py", line 239, in __call__
    self._handler.send(None)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 418, in handler
    await call_coroutine(entity, arg)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 332, in _execute_timer
    await await_or_execute(tmr.callback)
  File "/opt/ros/humble/local/lib/python3.10/dist-packages/rclpy/executors.py", line 107, in await_or_execute
    return callback(*args)
  File "/home/yyaker/pyroborsim_ws/install/pyrobosim_ros/lib/pyrobosim_ros/entrepot_pddl_planner.py", line 109, in timer_callback
    self.do_plan()
  File "/home/yyaker/pyroborsim_ws/install/pyrobosim_ros/lib/pyrobosim_ros/entrepot_pddl_planner.py", line 157, in do_plan
    plan_msg = task_plan_to_ros(plan)
  File "/home/yyaker/pyroborsim_ws/install/pyrobosim_ros/local/lib/python3.10/dist-packages/pyrobosim_ros/ros_conversions.py", line 208, in task_plan_to_ros
    raise Exception("Input is not a TaskPlan object")
Exception: Input is not a TaskPlan object
[ros2run]: Process exited with failure 1
```

Here is my planner file :

```python
`#!/usr/bin/env python3

"""
Test script showing how to use a PDDLStream planner as a ROS 2 node.
"""

[... 264 lines cut ...]
from pyrobosim.utils.general import get_data_folder
from pyrobosim_ros.ros_interface import WorldROSWrapper

def load_world():
    """Load a test world."""
    world_file = os.path.join(get_data_folder(), "entrepot_world.yaml")
    return WorldYamlLoader().from_yaml(world_file)

def create_ros_node():
    """Initializes ROS node"""
    rclpy.init()
    world = load_world()
    node = WorldROSWrapper(world=world, name="pddl_demo", state_pub_rate=0.1)
    return node

if __name__ == "__main__":
    node = create_ros_node()

    # Start ROS Node in separate thread
    ros_thread = threading.Thread(target=node.start)
    ros_thread.start()

    # Start GUI in main thread
    start_gui(node.world)
```

sea-bass, 2023-06-28:
Hi @yassineyakercap , I will take the action to at least not just throw an Exception is the task plan is none in the ROS interface.

I don't know what your world file looks like, but I suspect I know the issue. The `At` predicate only works with location names like `table0_tabletop`, but not with room names like `warehouse`... so that's probably failing to plan.

What you could do, if you want to enforce this, is use the `02_derived` planning domain and then use `("Has", "warehouse", "robot0")` as your goal predicate.

sea-bass, 2023-06-28:
Exception itself fixed in https://github.com/sea-bass/pyrobosim/pull/117

yassineyakercap, 2023-06-29:
Hello,

thanks for the help. I tried your idea to change the exemple parameter to 02_derived and then use ("Has", "warehouse", "robot0") as my goal predicate but my Task Plan was still None.

I guess I don't really understand how goal predicates works. My first goal was to test if I could make the lrobot go to table0 and pick up box_A0. Then it would go to shelf0 and place it there. But I can't even make it move to table0.

Here is my world YAML file : 

```yaml
params:
  name: entrepot_world
  object_radius: 0.1
  wall_height: 2.0

metadata:
  locations: $DATA/example_location_data.yaml
  objects: $DATA/example_object_data.yaml

robots:
  - name: lrobot
    radius: 0.1
    location: warehouse
    pose: [2.40, 2.0, 0.0]
    path_planner:
      type: rrt
      max_connection_dist: 0.5
      bidirectional: true
      rrt_star: true
      rewire_radius: 2.0
      compress_path: true
    path_executor:
      type: constant_velocity
      linear_velocity: 1.0
      max_angular_velocity: 4.0
      dt: 0.1

  - name: rrobot
    radius: 0.1
    location: warehouse
    pose: [3.60, 2.0, 0.0]
    path_planner:
      type: rrt
      max_connection_dist: 0.5
      bidirectional: true
      rrt_star: true
      rewire_radius: 2.0
      compress_path: true
    path_executor:
      type: constant_velocity
      linear_velocity: 1.0
      max_angular_velocity: 4.0
      dt: 0.1

rooms:
  - name: warehouse
    footprint:
      type: polygon
      coords:
        - [0, 0]
        - [6, 0]
        - [6, 4]
        - [0, 4]
    wall_width: 0.2
    color: [0.5, 0.5, 0.5]

locations:
  - category: table
    parent: warehouse
    pose: [1.0, 2.0, 0.0]

  - category: table
    parent: warehouse
    pose: [5.0, 2.0, 0.0]

  - category: shelf
    parent: warehouse
    pose: [3.0, 3.5, 0.0]

  - category: shelf
    parent: warehouse
    pose: [3.0, 1.0, 0.0]

objects:
  - category: box_A
    parent: table0

  - category: box_B
    parent: table1

```

And here is the exemple object and exemple location files that I have modified : 

```yaml
###########################
# Example object metadata #
###########################

apple:
  footprint:
    type: circle
    radius: 0.06
  color: [1, 0, 0]

banana:
  footprint:
    type: box
    dims: [0.05, 0.2]
  color: [0.7, 0.7, 0]

box_A:
  footprint:
    type: box
    dims: [0.2, 0.2]
  color: [0.64, 0.56, 0.48]

box_B:
  footprint:
    type: box
    dims: [0.2, 0.2]
  color: [0.64, 0.56, 0.48]

water:
  footprint:
    type: polygon
    coords:
      - [0.035, -0.075]
      - [0.035, 0.075]
      - [0.0, 0.1]
      - [-0.035, 0.075]
      - [-0.035, -0.075]
  color: [0.0, 0.1, 0.7]

coke:
  footprint:
    type: mesh
    model_path: $DATA/sample_models/coke_can
    mesh_path: meshes/coke_can.dae
  color: [0.8, 0, 0]

```

```yaml
#############################
# Example location metadata #
#############################

table:
  footprint:
    type: box
    dims: [0.9, 1.2]
    height: 0.5
  nav_poses:
    - [-0.75, 0, 0]
    - [0.75, 0, 3.14]
  locations:
    - name: "tabletop"
      footprint:
        type: parent
        padding: 0.1
  color: [0.2, 0, 0]

desk:
  footprint:
    type: polygon
    coords:
      - [-0.3, -0.3]
      - [0.3, -0.3]
      - [0.3, 0.3]
      - [-0.3, 0.3]
    height: 0.3
  locations:
    - name: "desktop"
      footprint:
        type: parent
      nav_poses:
        - [0, -0.5, 1.57]
        - [-0.5, 0, 0]
        - [0, 0.5, -1.57]
        - [0.5, 0, 3.14]

counter:
  footprint:
    type: box
    dims: [1.2, 0.6]
    height: 0.75
  locations:
    - name: "left"
      footprint:
        type: polygon
        coords:
          - [-0.25, -0.25]
          - [0.25, -0.25]
          - [0.25, 0.25]
          - [-0.25, 0.25]
        offset: [0.3, 0]
      nav_poses:
        - [0, 0.5, -1.57]
        - [0, -0.5, 1.57]
    - name: "right"
      footprint:
        type: polygon
        coords:
          - [-0.25, -0.25]
          - [0.25, -0.25]
          - [0.25, 0.25]
          - [-0.25, 0.25]
        offset: [-0.3, 0]
      nav_poses:
        - [0, 0.5, -1.57]
        - [0, -0.5, 1.57]
  color: [0, 0.2, 0]

trash_can:
  footprint:
    type: mesh
    model_path: $DATA/sample_models/first_2015_trash_can
    mesh_path: meshes/trash_can.dae
  locations:
    - name: "top"
      footprint:
        type: parent
        padding: 0.05
  nav_poses:
    - [0.5, 0.0, 3.14]
    - [-0.5, 0.0, 0.0]
  color: [0, 0.35, 0.2]

shelf:
  footprint:
    type: box
    dims: [1.0, 0.5]
    height: 1.0
  locations:
    - name: "shelf"
      footprint:
        type: parent
      nav_poses:
        - [0, -0.5, 1.57]
        - [0, 0.5, -1.57]
  color: [0.8, 0.8, 0.8]

```

sea-bass, 2023-06-30:
Seeing your world file, there is only one room... so the plan will be `None` because the robot can only be in that room.

Did you try `("At", "robot", "table_tabletop")` as a goal?

yassineyakercap, 2023-06-30:
It worked !!!

("At", "lrobot", "table0_tabletop"), did the job ! I guess you can't move to the same room you are in already.

Thanks a lot for the help.

Do you think you could have more informations about how to use the goal predicates in your documentation ?

sea-bass, 2023-06-30:
I guess I could add some more to https://pyrobosim.readthedocs.io/en/latest/usage/tamp.html on the goal predicates.

For reference, each PDDL domain will have its own predicates used for planning, so can't say much about that; however, there is a utility in pyrobosim that generates the **initial state's** predicates: https://github.com/sea-bass/pyrobosim/blob/main/pyrobosim/pyrobosim/planning/pddlstream/utils.py#L49

The key thing to know is:
* The `At` predicate is the actual thing that changes when you run these actions, and uses explicit instances that the robot moves to, like `table0_tabletop`
* The `Has`, `HasAll`, etc. predicates are **derived predicates** introduced in the [`02_derived` domain](https://github.com/sea-bass/pyrobosim/blob/main/pyrobosim/pyrobosim/data/pddlstream/domains/02_derived/domain.pddl#L88), and can be used to express higher-level concepts

yassineyakercap, 2023-07-03:
Thanks for the info

I'll close the issue now

closed 2023-07-03 by yassineyakercap