sea-bass/pyrobosim #127: Implement equality checkers for `Pose` class

sea-bass, 2023-07-23:
Several utilities and tests in `pyrobosim` rely on checking if poses are equal to each other.

Consider adding both
* An overloaded equality operator (the `__eq__()` method)
* A class function that checks if two poses are equal within certain position/orientation tolerances

ibrahiminfinite, 2023-07-23:
Such coincidence!
I was reading some python stuff yesterday regarding  object equality and was thinking about the Node comparisons, we had discussed it once before.

I think Nodes should also have a proper equality check.
Right now it uses the hashing to check equality.

sea-bass, 2023-07-23:
Yeah, I think for nodes it's a bit different because depending on the algorithm you may actually have two duplicate nodes with the same contents. In that sense, it's a true "Object", so it's more debatable whether we should do that.

For poses, it's definitely worthwhile to check that basically all the position/orientation values are the same even if it's two different objects.

JoshuaFutcher, 2023-07-24:
Hi, I can try sorting this : ) 
Would the tolerance apply to all of the values or groups of values?

PR #130 (2023-07-25): Adds Pose equivalency functions
Fixes #127 - Weird issue with rel_tol=1e-09 being Pose instead of float?
closed 2023-07-25 by sea-bass