sea-bass/pyrobosim #135: Use Trajectory class instead of tuples

sea-bass, 2023-08-20:
Right now, trajectories are represented as tuples of `(t_pts, x_pts, y_pts, yaw_pts)` which could be made nicer.

I propose doing something similar to what is in the `pyrobosim.utils.motion.Path` class, which contains a list of poses, but expanding this to include times as well.

ibrahiminfinite, 2023-09-04:
This one is a bit trickier than it seems.

There are 2 ways to go about this
If we store as poses, there will have to be 2 additional conversions on each call of the interpolation function.
1. Pose to numpy arrays
2. Numpy arrays to poses

The other way involves re-writing the interpolation function to do the interpolation manually while iterating over poses instead of using numpy. This will have more performance, but not sure if it will be significant enough.

Option 1 is easier and we can use the trusty numpy methods, but at the added cost of the 2 conversions.

ibrahiminfinite, 2023-09-06:
I actually implemented this strategy, but it felt very inefficient to convert numpy to Pose and back multiple time.
But looks like we don't have much choice here.

sea-bass, 2024-03-17:
> I actually implemented this strategy, but it felt very inefficient to convert numpy to Pose and back multiple time. But looks like we don't have much choice here.

You were right -- converting back and forth would have been inefficient, so I made the class a little more lightweight than this by internally storing the necessary `np.array`s

closed 2024-03-18 by sea-bass