sea-bass/pyrobosim #176: Separate ROS and Python packages

DLu, 2024-05-15:
Rather than doing the [full local setup](https://pyrobosim.readthedocs.io/en/latest/setup.html#local-setup) it would be nice to be able to do the `pip` install of the package, and then just clone the ROS packages to get them to work together. That would be a good first step for releasing the ROS packages as binaries.

scastro-bdai, 2024-05-15:
Yes, that would be nice now that there are actually "stable" releases of this tool.

The part that psyched me out about doing that is also not knowing how to make sure the pip dependencies of `pyrobosim` also get installed. Any ideas how to go about that? Because I couldn't figure it out / wasn't interested enough to dig further.

sea-bass, 2024-05-15:
Yes, that would be nice now that there are actually "stable" releases of this tool.

The part that psyched me out about doing that is also not knowing how to make sure the pip dependencies of pyrobosim also get installed. Any ideas how to go about that? Because I couldn't figure it out / wasn't interested enough to dig further.

sea-bass, 2024-08-24:
As of #246, at least now you can get around the `pip` install and actually run `colcon build` to install the non-ROS core package. So that's some progress...