sea-bass/pyrobosim #181: File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template       _interpreter = em.Interpreter(    AttributeError: module 'em' has no attribute 'Interpreter'

ct2034, 2024-06-28:
When following the setup instructions, https://pyrobosim.readthedocs.io/en/latest/setup.html and first setup the venv, then run `colcon build` I end up with this error:

<details><summary>console output</summary>
<p>

```
CMake Error at /opt/ros/humble/share/rosidl_adapter/cmake/rosidl_adapt_interfaces.cmake:59 (message):
  execute_process(/home/hec2le/python-virtualenvs/pyrobosim/bin/python3 -m
  rosidl_adapter --package-name pyrobosim_msgs --arguments-file
  /home/hec2le/ros/delib_ws/build/pyrobosim_msgs/rosidl_adapter__arguments__pyrobosim_msgs.json
  --output-dir
  /home/hec2le/ros/delib_ws/build/pyrobosim_msgs/rosidl_adapter/pyrobosim_msgs
  --output-file
  /home/hec2le/ros/delib_ws/build/pyrobosim_msgs/rosidl_adapter/pyrobosim_msgs.idls)
  returned error code 1:

  AttributeError processing template 'msg.idl.em'

  Traceback (most recent call last):

    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 48, in evaluate_template
      _interpreter = em.Interpreter(

  AttributeError: module 'em' has no attribute 'Interpreter'

  

  During handling of the above exception, another exception occurred:

  

  Traceback (most recent call last):

    File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
      exec(code, run_globals)
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__main__.py", line 19, in <module>
      sys.exit(main())
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/main.py", line 53, in main
      abs_idl_file = convert_to_idl(
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/__init__.py", line 19, in convert_to_idl
      return convert_msg_to_idl(
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/msg/__init__.py", line 39, in convert_msg_to_idl
      expand_template('msg.idl.em', data, output_file, encoding='iso-8859-1')
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 23, in expand_template
      content = evaluate_template(template_name, data)
    File "/opt/ros/humble/local/lib/python3.10/dist-packages/rosidl_adapter/resource/__init__.py", line 69, in evaluate_template
      _interpreter.shutdown()

  AttributeError: 'NoneType' object has no attribute 'shutdown'

Call Stack (most recent call first):
  /opt/ros/humble/share/rosidl_cmake/cmake/rosidl_generate_interfaces.cmake:130 (rosidl_adapt_interfaces)
  CMakeLists.txt:37 (rosidl_generate_interfaces)
``` 
</p>
</details> 

I was able to get around this by first building the workspace without venv. Recurrent build are okay with the venv.

Tested under humble / python 3.10

sea-bass, 2024-06-28:
Yeah, this is an annoying one...

it works if you first activate the venv and THEN do `source /opt/ros/humble/setup.bash` before building.

The convenience script attempts to do this in being able to call `source setup/source_pyrobosim.bash humble`.

Let me better document that in the local setup...

sea-bass, 2024-06-28:
Actually, I'm now noticing that is not enough. It has to do specifically with the `EmPy<4` requirement that ROS enforces.

Can you try installing `pip3 install colcon_common_extensions` in the venv to see if that works for you? If it does, I can figure out some updates.

ct2034, 2024-06-28:
No, got the same behavior. pip installed EmPy-3.3.4. But the behavior is the same

sea-bass, 2024-06-28:
I see... let's try something else. Could you:

1. Check out [this branch](https://github.com/sea-bass/pyrobosim/pull/183)
2. Set up with `source setup/create_python_env.bash ros` (the extra `ros` arg)
3. `source setup/source_pyrobosim.bash humble`
4. `colcon build`

Let me know...

ct2034, 2024-06-28:
Sorry I am running out of time. But this did not work:L
```
ERROR: Invalid requirement: './pyrobosim'
Hint: It looks like a path. File './pyrobosim' does not exist.
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'test/python_test_requirements.txt'
bash: [: missing `]'
```

ct2034, 2024-06-28:
Why do you need this venv? It makes it all a bit complicated to handle. ;)

sea-bass, 2024-06-28:
Yeah, I was running out of time myself. Will revisit this a bit later and make those scripts a little friendlier... but I think I understand the issue.

Sadly, it being first a standalone package and secondly a ROS wrapper creates these sorts of problems.

sea-bass, 2024-06-28:
> Why do you need this venv? It makes it all a bit complicated to handle. ;)

venvs are not needed, but they provide isolation. And in Ubuntu 24.04 you can't install directly into system Python, so we all have to move towards virtual envs or similar.

ct2034, 2024-06-28:
for python packages, yes. But they really don't play nicely with ros

sea-bass, 2024-06-28:
They indeed do not...

There is a related issue here to try address it by separating packages out.
https://github.com/sea-bass/pyrobosim/issues/176

The idea would be to add pyrobosim to rosdep and have the ROS wrapper package simply depend on it as a binary install.

Would be nice to do, but I definitely will need help to figure that out.

sea-bass, 2024-06-29:
@ct2034 I just took a more thorough pass at this problem and improved the setup scripts in https://github.com/sea-bass/pyrobosim/pull/183

I agree with you that the "right" thing for a ROS package is to make this work without virtual environments, but this is going to a big sweeping change that I am not able to do in the short term... nor do I want to necessarily make the ROS experience the primary goal of this package.

closed 2024-07-02 by sea-bass