sea-bass/pyrobosim #199: Add type hints

sea-bass, 2024-07-09:
I want to do this so badly! But if someone wants to help, I won't be mad.

MyPy is promising: https://mypy-lang.org/, which has pre-commit hooks at https://github.com/pre-commit/mirrors-mypy

kumar-sanjeeev, 2024-07-10:
Hi @sea-bass 

Can I work on this ?

sea-bass, 2024-07-11:
No need to ask! Please go ahead and submit a PR if/when you're ready. Thank you!

kumar-sanjeeev, 2024-07-19:
Hi @sea-bass 

I have a question. what is your recommendation? Should I submit a single PR for the whole repo or multiple PRs? For example, one PR for pyrobosim, another PR for pyrobosim_ros, and so on

sea-bass, 2024-07-19:
Good question.

I think the first order of business is to add mypy to the pre-commit check so it runs on CI:
https://github.com/pre-commit/mirrors-mypy

Unless this has a way to filter by folder, we may have to do the entire repo at once in order for CI to pass.

I would recommend starting with a draft PR that applies this for one file, because I am interested in also seeing what it ends up looking like and how we have to modify the docstrings / docs generation so the docs look good with type annotations.

kumar-sanjeeev, 2024-07-20:
Okay, I understand.

Yes, there is a way to filter folders/ files/directories to exclude them from the mypy check. This can be done by enabling or using the `--exclude` option from the configuration file.

https://mypy.readthedocs.io/en/stable/config_file.html

I created a similar config file in the root dir and excluded the major directories in the repository from the mypy check. Now, I am able to run this file from the command-line interface like this:

```
mypy --config-file "CONFIG_FILE_NAME" .
```
This is running fine.

Now, because we want to configure mypy with pre-commit, I tried to read the config file from the pre-commit hook in the form of an argument. As of now, it is not working. I am figuring this out.

```
-   repo: https://github.com/pre-commit/mirrors-mypy
    rev: ''v1.10.1 " 
    hooks:
    -   id: mypy
        args: [--config-file, CONFIG_FILE_NAME]
```

As you suggested, I will do the first draft PR that applies type hints to only one file, and I will exclude the rest of the files using the `--exclude` option.

sea-bass, 2025-02-14:
@kumar-sanjeeev Just in case, I'm now working on this in #329

PR #329 (2025-02-24): Add mypy and type annotations 🚀 
This is THE type hinting PR. Gonna be a big one...

In addition to just making mypy happy, I've done a bunch of refactoring / reshuffling files -- mostly based on circular imports due to importing types for hinting.

So this means the next release will be a 4.0.

Closes #199
closed 2025-02-24 by sea-bass