sea-bass/pyrobosim #200: Add UI button to toggle collision polygon display

sea-bass, 2024-07-09:
There already is a hard-coded flag that shows the locations' collision polygons for debugging.

Would be nice if this were formally a button or checkbox, maybe below the MatPlotLib canvas.

muhidabid, 2024-08-17:
Hi, I'd like to work on this. Can you please point out what are the "locations' collision polygons" in this screenshot?

Which files should I look into to implement this? Any tips on understanding the code base?

[image]

sea-bass, 2024-08-18:
If you look at this line and set it to True, you will see the collision polygons displayed: https://github.com/sea-bass/pyrobosim/blob/ea926bbdf736f5dbc2537dcc0d35789bb54e2d86/pyrobosim/pyrobosim/gui/world_canvas.py#L153

The idea is to change that from being hard-coded to being a toggle display. Some useful things to look for:

* The file I linked above has all the visualization canvas, but the GUI definition and buttons are here: https://github.com/sea-bass/pyrobosim/blob/main/pyrobosim/pyrobosim/gui/main.py
* I assume the matplotlib `set_visible()` method of artists will come in handy: https://matplotlib.org/stable/api/_as_gen/matplotlib.artist.Artist.set_visible.html

sea-bass, 2024-08-19:
Closes by #242

closed 2024-08-19 by sea-bass