sea-bass/pyrobosim #215: Allow canceling actions from the GUI

sea-bass, 2024-07-14:
Since #214 was merged in, we now have all the backend APIs to do this, either for single actions that were kicked off by pushing GUI buttons, or because something in the backend sent an action or task plan to the robot(s).

The "hard" part is just figuring out where in the GUI to put this cancel button.

The `test/system/test_system.py` file should probably also have a test for clicking this cancel button.

muhidabid, 2024-08-20:
I am thinking the cancel button could be on the bottom right corner of the world, opposite to the "show collision polygons" checkbox. We can print a short message for the actions letting the user know something is in progress along with the cancel button. What do you think?

sea-bass, 2024-08-20:
Was thinking something like that,  especially since I think after one of the new PRs there may also be a need for a "reset path planner" button as well

muhidabid, 2024-08-20:
Maybe an `undo` would be nice instead of different implementations of `cancel` and `reset path planner`. You can undo any of your actions once they are done. Just thinking out loud

sea-bass, 2024-08-20:
Undo would require some significant backend changes (and I at least don't see a use case for it right now).

Canceling a currently executing path and resetting path planners are already implemented.

muhidabid, 2024-08-20:
Alright. I will work on this issue. Should I also look into reset path planner or will there be a separate issue for that? 

I was looking at the commit you made in #214. Is this the API to be called upon cancel? https://github.com/sea-bass/pyrobosim/blob/main/pyrobosim/pyrobosim/core/robot.py#L903

sea-bass, 2024-08-20:
Yep, that's the one!

And let's save the reset path planner til later since that PR is still waiting for review

muhidabid, 2024-08-21:
@sea-bass how does this layout look?

[image]

closed 2024-08-24 by sea-bass