sea-bass/pyrobosim #236: error running 'python3 examples/demo.py'

muhidabid, 2024-08-12:
I'm trying to standalone set up pyrobosim following the [docs](https://pyrobosim.readthedocs.io/en/latest/usage/basic_usage.html). I thought I had successfully set it up but upon running the demo I get the following error: [error gist](https://gist.github.com/muhidabid/abd9c2821af60bc11991bdf2fdfa58e5)

sea-bass, 2024-08-12:
Searching for this error message online, I see this answer: https://stackoverflow.com/a/74113918

What version of Python and/or PySide6 are you using?

Also, most importantly, the error message indicates you are using your system Python instead of a virtual environment. Have you tried going through the setup scripts in the docs that make and use a virtual environment, to see if that helps keep the dependencies in check?

sea-bass, 2024-08-14:
Just to be safe, I'm putting a minimum of 6.4.0 on the PySide6 version according to the previous Stack Overflow answer.

See https://github.com/sea-bass/pyrobosim/pull/238

If you can confirm whether this fixes the issue, I would appreciate it.

muhidabid, 2024-08-14:
I was missing `sudo apt install python3.10-venv`. Installing that then going through the setup fixed this issue. Thank you

closed 2024-08-14 by muhidabid