sea-bass/pyrobosim #252: Closing the GUI when ROS is up produces a stack trace before shutting down

sea-bass, 2024-08-27:
If you launch the basic ROS demo:

```
ros2 launch pyrobosim_ros demo.launch.py
```

Simply closing the GUI with the "X" button yields this stack trace:

```
[demo.py-1] Exception in thread Thread-1 (<lambda>):
[demo.py-1] Traceback (most recent call last):
[demo.py-1]   File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
[demo.py-1]     self.run()
[demo.py-1]   File "/usr/lib/python3.12/threading.py", line 1010, in run
[demo.py-1]     self._target(*self._args, **self._kwargs)
[demo.py-1]   File "/home/sebastian/workspace/pyrobosim_ws/install/pyrobosim_ros/lib/pyrobosim_ros/demo.py", line 135, in <lambda>
[demo.py-1]     ros_thread = threading.Thread(target=lambda: node.start(wait_for_gui=True))
[demo.py-1]                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[demo.py-1]   File "/home/sebastian/workspace/pyrobosim_ws/install/pyrobosim_ros/lib/python3.12/site-packages/pyrobosim_ros/ros_interface.py", line 199, in start
[demo.py-1]     executor.spin()
[demo.py-1]   File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/executors.py", line 311, in spin
[demo.py-1]     self.spin_once()
[demo.py-1]   File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/executors.py", line 902, in spin_once
[demo.py-1]     self._spin_once_impl(timeout_sec)
[demo.py-1]   File "/opt/ros/rolling/lib/python3.12/site-packages/rclpy/executors.py", line 892, in _spin_once_impl
[demo.py-1]     self._executor.submit(handler)
[demo.py-1]   File "/usr/lib/python3.12/concurrent/futures/thread.py", line 170, in submit
[demo.py-1]     raise RuntimeError('cannot schedule new futures after shutdown')
[demo.py-1] RuntimeError: cannot schedule new futures after shutdown
[demo.py-1] sys:1: RuntimeWarning: coroutine 'Executor._make_handler.<locals>.handler' was never awaited
[demo.py-1] RuntimeWarning: Enable tracemalloc to get the object allocation traceback
[INFO] [demo.py-1]: process has finished cleanly [pid 130515]
```

Nothing is wrong, per se, but it would be nice to find a way to shut down more cleanly than this.

dcconner, 2024-08-31:
Simple except addition at line 200 in ros_interface.py will at least clean up the message, even without catching the X and shutting down completely clean.
```
            try:
                executor.spin()
            except Exception as exc:
                print(exc)
            finally:
                executor.remove_node(self)
                self.destroy_node()
                executor.shutdown()
                rclpy.shutdown()
```
 e.g.

```
~/iron$ ros2 run delib_ws_p1 run
[INFO] [1725134912.935020374] [pyrobosim]: World node started.
[INFO] [1725134913.124636506] [pyrobosim]: PyRoboSim ROS node ready!
cannot schedule new futures after shutdown
sys:1: RuntimeWarning: coroutine 'Executor._make_handler.<locals>.handler' was never awaited
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
```

sea-bass, 2024-08-31:
Thanks. Actually, I found a more principled fix -- to include the shutdown as part of the GUI's close event :)

https://github.com/sea-bass/pyrobosim/pull/256 works great on my end.

PR #256 (2024-08-31): Cleanly shutdown ROS interface node
This PR cleanly shuts down the ROS interface node by bringing down the ROS executor as part of the GUI's close event, and handling some other logic a bit.

Closes #252
closed 2024-08-31 by sea-bass