sea-bass/pyrobosim #272: Allow path planning and execution with partial observability of hallway open/closed state

sea-bass, 2024-09-17:
Just as we have robots with partial observability of objects, it would be useful to do the same for path planning.

The use case is:
* Plan a path through a specific hallway
* Realize that the hallway is actually closed (requires some notion of a local collision detector with finite radius)
* Path execution fails
* (Optionally) information about the hallway is updated

WTTien, 2025-03-14:
Hi, I am quite interested to implement this feature. 

From my understanding, in execution.py, a thread to runs validate_remaining_path every 0.5s. At the same time, the interpolated trajectory is computed before actually starting the robot executions (presumably through self.robot.set_pose(cur_pose)(?)).

I am curious about the significance of running validate_remaining_path every 0.5s, when the collisions along trajectory can be analysed once before executing it?
(Could this be used to accomodate scenarios when hallway state changes throughout execution?)

sea-bass, 2025-03-14:
> Could this be used to accomodate scenarios when hallway state changes throughout execution?)

That was exactly it!

Also at some point I'd like to add a simulated lidar sensor to detect obstacles generally.

WTTien, 2025-03-14:
Nice, thanks for the info! 

Do you mind if I implement some sort of sensor modules for the robot as part of this enhancement? Or would it be better for it to be in a different exercise?

WTTien, 2025-04-04:
Hi I created a partial_observability_hallway_states parameter. By setting it to true the robot would plan its path just as if all hallways are open (for now). 

I also created a simple lidar module that scans around the robot at certain radius whenever path execution starts. Just like validate_remaining_path it runs on a thread and raises the abort_execution flag whenever an occupied pose is detected.

There is still a lot more work done but I just wanted to get some input if this is a good direction..

(Additionally I created an artist to draw the lidar scanning range. It interferes with the robot movement artist for now but I'm sure it can be sorted)

I did my work here: https://github.com/WTTien/pyrobosim_hallway_partial_observalibility

sea-bass, 2025-04-04:
Awesome @WTTien !

I have also been working on a lidar sensor implementation, and the whole sensor model/background thread is very similar. However, I think I have an implementation that might be faster since it doesn't require sampling step distances.

I will try clean this up when I find time and share it with you here. I still think my implementation is too slow, which is why I gave up on it, but maybe still worth sharing for now.

WTTien, 2025-04-05:
Sounds good thanks!

sea-bass, 2025-04-06:
@WTTien here is my draft branch -- I still have a lot of work to do, but maybe it's useful?

https://github.com/sea-bass/pyrobosim/pull/344

WTTien, 2025-04-08:
Looks very useful! I'll implement them 😄

WTTien, 2025-05-25:
I think this is worth a try now
https://github.com/sea-bass/pyrobosim/pull/357 

We can run it using python pyrobosim/examples/demo.py --partial-observability-hallway-states

closed 2025-07-04 by sea-bass