sea-bass/pyrobosim #299: Allow supporting multiple sources of object/location metadata

sea-bass, 2024-11-09:
Another great idea suggested by a user.

This would allow not needing to copy-paste as much, and under the hood it's really just a bunch of dictionary merges.

kumar-sanjeeev, 2024-12-18:
Hi @sea-bass, I would like to work on this issue, but I’m not entirely sure what needs to be done here. From what I see in the current codebase, location and object metadata are defined in YAML files and loaded using `world.set_metadata`, either directly through the `World` class or via the `WorldYamlLoader`. Could you share more details about what you have in mind?

sea-bass, 2024-12-18:
Correct -- so right now, you can only link to a single metadata file.

But what if you have multiple file sources? For example, you may have:
* Food objects
* Drink objects
* Garbage objects
* ... etc

What if you could specify multiple sources of metadata, so instead of `set_metadata()` there was something like `add_metadata()` that would combine knowledge from different files to prevent copy-paste?

kumar-sanjeeev, 2024-12-23:
Hi @sea-bass, thanks for explaining. With this much information, I’ll start thinking and get back to you once everything is clear in my mind.

kumar-sanjeeev, 2024-12-24:
I did some digging and planning to implement something like this:

**Current Codebase:**

`World.set_metadata` currently accepts a single source of metadata for `locations` and `objects`. Under the hood `World` class is delegating the set_metadata responsibility to individual `Locaton.set_metadata` and `Object.set_metadata`, which further call `EntityMetadata` to load the parsed metadata (YAML) into a dict. 

During this process, `EntityMetadata` saves the filename, which later used by `WorldYamlWriter`. Since the current implementation uses a single metadata source, the `self.filename` attribute in `EntityMetadata` is enough to store information 

**Approach for supporting multi sources of locations/objects metadata:**

As you suggested, we can define `add_metadata` something, so I am planning to add this method to the `World`, `Location`, and `Object` classes, similar to how `set_metadata` is used in the current implementation. The `add_metadata` methods in Location and Object will call an `update` method in `EntityMetadata`, which will be responsible for `merging metadata from different sources`.

For example:

```
class World:
   ....
   def add_metadata(self, locations=None, objects=None):

        if locations is not None:
            Location.add_metadata(locations)
        if objects is not None:
            Object.add_metadata(objects)
   ....

class Location:
   ....
   @classmethod
    def add_metadata(cls, filename):
        new_metadata = EntityMetadata(filename)
        cls.metadata.data.update(new_metadata.data)
   ....

class Object:
   ....
   @classmethod
    def add_metadata(cls, filename):
        new_metadata = EntityMetadata(filename)
        cls.metadata.data.update(new_metadata.data)
   ....

class EntityMetadata :
  self.data
   ....
   def update(self, new_data):
        self.data.update(new_data)
   ....
```

Based on above implementation (rough idea) we can create a world from different sources like this:

```
world.add_metadata(
        locations=os.path.join(data_folder, "location1.yaml"),
        objects=os.path.join(data_folder, "object1.yaml")
    )

 world.add_metadata(
      locations=os.path.join(data_folder, "location2.yaml"),
      objects=os.path.join(data_folder, "object2.yaml"),
  )

 world.add_metadata(
      locations=os.path.join(data_folder, "location3.yaml"),

```

One challenge I foresee with this approach is ensuring that I collect all the source files that define metadata for `locations` and `objects`. Based on this, I would need to modify the logic in `WorldYamlWriter`, which is especially important when the `reset-world` option is selected in the GUI.

What do you think of this approach?

kumar-sanjeeev, 2024-12-24:
FYI, today I tried implementing the approach shared above, and it is working fine so far. I am planning to test it thoroughly from every possible angle.Once I am confident it is working well and after receiving your feedback on the shared approach, I will raise a `draft PR`, so that you can check out the working model. If any changes or additions/deletions are needed, I’ll address them there.

kumar-sanjeeev, 2024-12-26:
Hi @sea-bass, does the above-shared approach align with your thoughts on supporting multiple sources of object/location metadata?

sea-bass, 2024-12-26:
> Hi @sea-bass, does the above-shared approach align with your thoughts on supporting multiple sources of object/location metadata?

Yes, this looks good! Feel free to put up a draft PR whenever you have something, and I'll look at the code.

closed 2025-01-06 by sea-bass