sea-bass/pyrobosim #301: Allow color specification in non-RGB formats

sea-bass, 2024-11-09:
The suggestion was "Would be nice to be able to specify colors as strings supporting standard color strings like "red" or "#0F0F0F""

There are probably some standard modules that can take care of this already, so it's a matter of searching.

kumar-sanjeeev, 2024-12-01:
Hi @sea-bass 

I did some search and my suggestion is to use [`matplotlib-specifying colors`](https://matplotlib.org/3.8.4/gallery/color/named_colors.html) functionality. It provides three classes of colors - `Base`, `Tableau Palette` and `CSS`. Since matplotlib is already a dependency of pyrobosim, this approach might make it easier to implement the required changes.

Let me know your thoughts. If this approach makes sense, I would like to contribute to this issue.

I tested my suggestion using the `demo.py` script, and here are the results:

## Code Changes:
```
import matplotlib.colors as mcolors

....

def create_world(multirobot=False):
    """Create a test world"""
    world = World()

    # Set the location and object metadata
    world.set_metadata(
        locations=os.path.join(data_folder, "example_location_data.yaml"),
        objects=os.path.join(data_folder, "example_object_data.yaml"),
    )

    # Add rooms
    r1coords = [(-1, -1), (1.5, -1), (1.5, 1.5), (0.5, 1.5)]
    world.add_room(
        name="kitchen",
        footprint=r1coords,
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:brown"]),
        nav_poses=[Pose(x=0.75, y=0.75, z=0.0, yaw=0.0)],
    )
    r2coords = [(1.75, 2.5), (3.5, 2.5), (3.5, 4), (1.75, 4)]
    world.add_room(
        name="bedroom",
        footprint=r2coords,
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:cyan"]),
    )
    r3coords = [(-1, 1), (-1, 3.5), (-3.0, 3.5), (-2.5, 1)]
    world.add_room(
        name="bathroom",
        footprint=r3coords,
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:pink"]),
    )

    # Add hallways between the rooms
    world.add_hallway(
        room_start="kitchen",
        room_end="bathroom",
        width=0.7,
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:olive"]),
    )
    world.add_hallway(
        room_start="bathroom",
        room_end="bedroom",
        width=0.5,
        conn_method="angle",
        conn_angle=0,
        offset=0.8,
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:olive"]),
    )
    world.add_hallway(
        room_start="kitchen",
        room_end="bedroom",
        width=0.6,
        conn_method="points",
        conn_points=[(1.0, 0.5), (2.5, 0.5), (2.5, 3.0)],
        color=mcolors.to_rgb(mcolors.TABLEAU_COLORS["tab:olive"]),
    )

     ....
```

## Results
[image]

kumar-sanjeeev, 2024-12-01:
Okay, understood. I just need one clarification: From what I understand, I need to change any [R, G, B] color values used in source code and YAML files with matplotlib color functionality, while keeping the values the same or similar. Is that correct? If I’m missing something, please let me know.

sea-bass, 2024-12-01:
Yeah, probably just if statements on whether the value is a list of numbers or a string...

kumar-sanjeeev, 2024-12-01:
okay cool, so I will start implementing with base colors as shown in this image.

```
{
  'b': (0, 0, 1), 
  'g': (0, 0.5, 0), 
  'r': (1, 0, 0), 
  'c': (0, 0.75, 0.75), 
  'm': (0.75, 0, 0.75), 
  'y': (0.75, 0.75, 0), 
  'k': (0, 0, 0),
  'w': (1, 1, 1)
}
```

[image]

sea-bass, 2024-12-07:
@kumar-sanjeeev knocked this one out! 🏅

closed 2024-12-07 by sea-bass