sea-bass/pyrobosim #310: ROS 2 action clients don't receive cancellation response

robin-mueller, 2024-12-19:
The [`WorldROSWrapper`](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L48) exposes actions and services to the ROS 2 network and makes it possible to request actions or query services. In the case of ROS 2 actions I've found, that the individual `rclpy.action.ActionServer` instances might not be implemented correctly, since action clients are not able to receive a response for a cancellation request.

### How to reproduce
1. Start the GUI and the default word running `demo.py`.
2. Send a simple `Navigate` action request. This is can be done using the terminal:

    ```sh
    ros2 action send_goal /execute_action pyrobosim_msgs/action/ExecuteTaskAction "{action: {type: navigate, robot: robot, target_location: bathroom}}"
    ```

3. Send a cancellation request by interrupting the terminal process pressing `Ctrl+C`

### Expected output
Usually, the program would print something like this if the cancellation request was sent and the corresponding response received correctly:

```
Waiting for an action server to become available...
Sending goal:
     action:
  robot: robot
  type: navigate
  object: ''
  room: ''
  source_location: ''
  target_location: bathroom
  cost: 0.0
  has_pose: false
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  path:
    poses: []
    length: 0.0

Goal accepted with ID: 32789fe07b644411b0611d1f0b7e6f02

Canceling goal...
Goal canceled.  # <-- This should appear if the cancellation request was accepted and the client received the response.
```
### Actual output
Instead, I receive an error message immediately after I requested to cancel with `Ctrl+C`.
```
Waiting for an action server to become available...
Sending goal:
     action:
  robot: robot
  type: navigate
  object: ''
  room: ''
  source_location: ''
  target_location: bathroom
  cost: 0.0
  has_pose: false
  pose:
    position:
      x: 0.0
      y: 0.0
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
  path:
    poses: []
    length: 0.0

Goal accepted with ID: 32789fe07b644411b0611d1f0b7e6f02

Canceling goal...
Exception while canceling goal: None  # <-- Instead, one observes this
```

Refer to [ros2cli/ros2action](https://github.com/ros2/ros2cli/blob/35d6abd50a55181099aab82134fc507d8fd162bd/ros2action/ros2action/verb/send_goal.py#L89) for more information about the methods that are called during a `ros2 action send_goal` command.

robin-mueller, 2024-12-19:
After some investigation I suspect the following:

Looking at the source code and the `action_server` instance that is responsible for handling `Navigate` action requests in particular, there is a small caveat with the corresponding [callback functions](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L384-L440). My thoughts are based on the assumption that with `rclpy` the action server callbacks are executed concurrently. Therefore, when a cancel request arrives, [`action_callback`](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L384) and [`action_cancel_callback`](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L429) are both called in the same iteration (NOT simultaneously but concurrently!). However, the execution order is unkown (it's depending on the operating systems scheduling logic). So there is kind of a race condition between [`robot.execute_action`](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L418) and [`robot.cancel_actions`](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L439). Both callbacks observe the `self.executing_nav` flag for determining when the execution has finished (See [here](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim/pyrobosim/core/robot.py#L906) and [here](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim/pyrobosim/core/robot.py#L952)).

In the end, it's impossible to know which callback observes that the flag is set to `False` first and purely determined by chance. **And that is the crucial point:** If `robot.execute_action` becomes aware of the execution completion before `robot.cancel_actions`, the server's execute callback `action_callback` will proceed and "succeed" the goal handle before the cancellation request has even been answered. When `action_cancel_callback` finally becomes aware of the current state, the goal handle has already reached a terminal state and an error occurs.

In my opinion, the action servers of `WorldROSWrapper` should be implemented according to the design that we apply with `rclcpp` (the C++ API). Meaning that the actual cancellation process should also be handled by the action's execute callback. Afterall, the intended responsibility of the cancel callback is only determining if the goal is allowed to be canceled or not. It's not supposed to do any real work.

sea-bass, 2024-12-19:
Thanks for raising this!

It might be that it seems the action callbacks always do a `goal_handle.succeed()` instead of a `goal_handle.canceled()` in the right conditions.

... so I think it's just checking whether the robot's action completed or was canceled.

I'll look into this when I have the chance.

sea-bass, 2024-12-19:
> Afterall, the intended responsibility of the cancel callback is only determining if the goal is allowed to be canceled or not. It's not supposed to do any real work.

Correct. And it does this currently. The cancel callbacks should tell the robot to cancel its running action, but the return status of that action is checked in the actual execution.

robin-mueller, 2024-12-19:
> It might be that it seems the action callbacks always do a goal_handle.succeed() instead of a goal_handle.canceled() in the right conditions.

That's what I thought as well, but according to the [action design document](https://design.ros2.org/articles/actions.html) it's totally fine to succeed when canceling.

> Correct. And it does this currently. The cancel callbacks should tell the robot to cancel its running action, but the return status of that action is checked in the actual execution.

Hmm, but what about [this line](https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim/pyrobosim/core/robot.py#L952) inside `robot.cancel_actions`. It seems to me that the method waits until `self.executing_nav` is `False`.

I'm not used to `rclpy` since I mainly work with `rclcpp`, so please forgive me if I'm wrong :)

sea-bass, 2024-12-19:
Yes, what happens is that this library's core is not based on ROS, so all actions return an `ExecutionResult` with them.

So in the case that an action is canceled, the navigation action still completes, but returns an `ExecutionStatus.Canceled`: https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim/pyrobosim/navigation/execution.py#L134-L139

So we need to check the return status of the execution result here: https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim_ros/pyrobosim_ros/ros_interface.py#L418 -- and appropriately check that to know whether we should call `goal_handle.succeed()` vs. `goal_handle.canceled()`.

robin-mueller, 2024-12-19:
Well I'd suggest using the execute callback for that, since it's still being called continuously after the cancellation request has been accepted. That's how I always do it with `rclcpp`. But that would require refactoring `action_callback` to not use `robot.execute_action` since this is also a blocking function. Or we merge the `cancel_actions` function into `execute_action` and let `cancel_action` only set some flag.

sea-bass, 2024-12-19:
Oh, another fun update. If I run that ROS 2 example, I also get the same "Exception when canceling goal: None"

```
ros2 run examples_rclpy_minimal_action_server server_single_goal 

ros2 action send_goal /fibonacci example_interfaces/action/Fibonacci "{order: 100}"

Ctrl + C
```

... so maybe this is an `rclpy` thing?

robin-mueller, 2024-12-19:
Oof.. it also seems to me that it should work. Just note, that there should probably also be some more logic that also determines when a cancel request should be rejected similar to this

https://github.com/sea-bass/pyrobosim/blob/1d356945c59c2ebaaa1e297d94dea5dc84ccc603/pyrobosim/pyrobosim/core/robot.py#L944-L947

sea-bass, 2024-12-19:
Good idea -- https://github.com/sea-bass/pyrobosim/pull/311/commits/55293eb853c06ab9a5a95a967adfb0ab72d2e7d5

robin-mueller, 2024-12-19:
Small update: I just cloned your pull request #311 and my application (the real one, not the CLI) now works correctly. My client is now able to successfully receive the goal request and conclude the cancellation on its side.

Thanks for your time! I guess there is still an issue with rclpy or ros2cli but let's see...

PS: Great work with pyrobosim! It's awesome to have a lightweight platform for behavior prototyping and demonstration. I'm currently integrating this project in one of my repo's to allow users to quickly validate their programs and I'm really happy to have come across pyrobosim

sea-bass, 2024-12-19:
Just merged. Thanks again for bringing this up.

I also look forward to seeing what you end up doing on your repo. If you have any usage examples to share, you can add them to https://github.com/sea-bass/pyrobosim/blob/main/docs/source/index.rst#usage-examples

And generally, let me know as anything else comes up.

closed 2024-12-19 by sea-bass