sea-bass/pyrobosim #319: Feature Request: Service to reset the world

SGeetansh, 2025-01-14:
I believe currently, the only way to reset the world is via the 'Reset world' button on the GUI. A ros2 service to reset the world from another ros2 node would be nice.

sea-bass, 2025-01-16:
This is a good idea -- right now the reset logic is embedded in the GUI code:
https://github.com/sea-bass/pyrobosim/blob/511c106b3a668e5b630eb8e8f5d30c81630e8fd0/pyrobosim/pyrobosim/gui/main.py#L395

However, it would be good to have a standalone function like `World.reset()`.

This could be called as-is, plus as you suggested a wrapper service for the ROS interface.

kumar-sanjeeev, 2025-01-16:
@sea-bass, would like to take this one.

sea-bass, 2025-02-27:
Hi @SGeetansh -- I have a working implementation here: https://github.com/sea-bass/pyrobosim/pull/333

However, note that the previous PR added type annotations and did a major refactor in the process. So there may be some incompatibilities depending on how you're using PyRoboSim.

Let me know how it goes!

PR #333 (2025-03-05): Refactor world resetting to allow standalone use and ROS service
This PR takes over from #321 to refactor world resetting so that it's usable in all modes of operation: standalone, through ROS service, and through the GUI.

Closes #319
closed 2025-03-05 by sea-bass