sea-bass/pyrobosim #324: Defining Hallway through footprint

PAyush15, 2025-01-27:
[image]

In the above image, I have specified the locations of rooms according to the floor plan and created them through APIs. Now, I want the hallway as I mentioned the area through black lines, such that it connects multiple rooms at once and also extend in some nearby areas. 

I tried creating it through "points" method for 'Hallways' but it seems difficult since the width of the hallway is different in different areas and the rooms are not so directly connected. Using the "points" method, the width of the hallway remains constant (also the width of the hallway is the width of the door/entrance, which is not exactly in my case). So, I cannot directly use it.

Is there another way I can create this type of hallway, maybe by defining it through footprint and specifying which room it connects?

sea-bass, 2025-01-27:
Hmmm, I think you might be best off defining this "Hallway" as an actual room, and using the PyRoboSim hallways to describe only the doors between the offices and the room using the "points" method.

PAyush15, 2025-01-29:
Okay. Yes, I think that would work well for my use-case. Thanks!