sea-bass/pyrobosim #325: IndexError: index 1 is out of bounds for axis 0 with size 0

PAyush15, 2025-02-03:
Hello all! I am trying to create a world from a floor-plan. The rooms seems good. Also, I created the hallways as rooms due to complex shape of the hallway. But when I try to combine different hallways, I get an error: `IndexError: index 1 is out of bounds for axis 0 with size 0. `

[image]

So for some examples it worked. Example, I was able to connect 'Hallway_1 and Hallway_2',  'Hallway_2 and Hallway_4' and 'Hallway_1 and Hallway_5'. But when I try to create a hallway between say Hallways 2 & 3 or Hallways 4 & 5 (or the remaining ones I have not connected, it gives me the following error: 

```
Traceback (most recent call last):
  File "/home/nitro/ros2_ws/src/pyrobosim/pyrobosim/examples/idac_world.py", line 414, in <module>
    world = create_world(args.multirobot)
  File "/home/nitro/ros2_ws/src/pyrobosim/pyrobosim/examples/idac_world.py", line 278, in create_world
    world.add_hallway(
  File "/home/nitro/ros2_ws/install/pyrobosim/lib/python3.10/site-packages/pyrobosim/core/world.py", line 272, in add_hallway
    hallway = Hallway(**hallway_config)
  File "/home/nitro/ros2_ws/install/pyrobosim/lib/python3.10/site-packages/pyrobosim/core/hallway.py", line 122, in __init__
    self.update_collision_polygons()
  File "/home/nitro/ros2_ws/install/pyrobosim/lib/python3.10/site-packages/pyrobosim/core/hallway.py", line 164, in update_collision_polygons
    self.inflated_closed_polygon = inflate_polygon(
  File "/home/nitro/ros2_ws/install/pyrobosim/lib/python3.10/site-packages/pyrobosim/utils/polygon.py", line 98, in inflate_polygon
    return orient(inflated_poly)
  File "/home/nitro/.local/lib/python3.10/site-packages/shapely/geometry/polygon.py", line 346, in orient
    if signed_area(ring) / s >= 0.0:
  File "/home/nitro/.local/lib/python3.10/site-packages/shapely/algorithms/cga.py", line 11, in signed_area
    xs, ys = np.vstack([coords, coords[1]]).T
IndexError: index 1 is out of bounds for axis 0 with size 0
```

Also, when I try to create a door using hallways to the rooms, it shows the same error. I checked multiple times, but the coordinates seems correct as I have kept the constant 0.1 wall width for all. But still the error seems to persist. More confusing to me is my it happens for some hallways and not for others? 

Can anyone please help me with this issue? Been on it since 3-4 days. Below is my code for the world as a reference. Thanks much!

```
#!/usr/bin/env python3

"""
Test script showing how to build a world and use it with pyrobosim
"""
import os
import argparse

from pyrobosim.core import Robot, World, WorldYamlLoader
from pyrobosim.gui import start_gui
from pyrobosim.manipulation import GraspGenerator, ParallelGraspProperties
from pyrobosim.navigation import (
    ConstantVelocityExecutor,
    AStarPlanner,
    PRMPlanner,
    RRTPlanner,
)
from pyrobosim.utils.general import get_data_folder
from pyrobosim.utils.pose import Pose

from shapely.geometry import Polygon

data_folder = get_data_folder()

def create_world(multirobot=False):
    """Create a test world"""
    world = World()

    # Set the location and object metadata
    world.add_metadata(
        locations=[
            os.path.join(data_folder, "example_location_data_furniture.yaml"),
            os.path.join(data_folder, "example_location_data_accessories.yaml"),
        ],
        objects=[
            os.path.join(data_folder, "example_object_data_food.yaml"),
            os.path.join(data_folder, "example_object_data_drink.yaml"),
        ],
[... 316 lines cut ...]
        "--world-file",
        default="",
        help="YAML file name (should be in the pyrobosim/data folder). "
        + "If not specified, a world will be created programmatically.",
    )
    parser.add_argument(
        "--partial-observability",
        action="store_true",
        help="If True, robots have partial observability and must detect objects.",
    )
    return parser.parse_args()

if __name__ == "__main__":
    args = parse_args()

    # Create a world or load it from file.
    if args.world_file == "":
        world = create_world(args.multirobot)
    else:
        world = create_world_from_yaml(args.world_file)

    # Start the program either as ROS node or standalone.
    start_gui(world)

```

sea-bass, 2025-02-04:
Hey @PAyush15 -- thanks for flagging this! I have good news for you.

For the first one that errors, this was an issue in PyRoboSim in which we're trying to inflate an empty polygon because there is zero room between the two walls. I've handled this case here: https://github.com/sea-bass/pyrobosim/pull/328

The second case (the one you commented out in your script) is because the hallway width is `0.1` which is the same size as the inflation radius, so this errors for that different reason! If you increase this hallway's width to `1.0` or so, it works fine.

[image]

closed 2025-02-04 by sea-bass

PAyush15, 2025-02-04:
Hello @sea-bass, thanks for the quick update! Works great now. It failed for me too with RRT while navigating due to the reason you mentioned. Maybe I'll end up using some other planner. But thanks for the heads up.

Just for curiosity, is the 'inflation radius' set to a fixed value (0.1) or we define it somewhere?

sea-bass, 2025-02-04:
The default inflation radius of worlds is zero, but then it gets modified when you add robots.

It so happens that the robot in this example has radius 0.1, so it ends up inflating the world by that radius!

PAyush15, 2025-02-06:
Got it. Thanks for the explanation!

PR #328 (2025-02-04): Handle empty polygons in inflate_polygon
This PR handles a few edge cases due to infinitely thin hallways in two dimensions, since Shapely doesn't seem to handle them gracefully:

1. No clearance between the walls of the two neighboring rooms (requires checking polygon emptiness before inflating)
2. Hallway half-width smaller than the world inflation radius (requires checking these values and returning an empty Polygon)

I think in both cases, I've narrowed it down to this: https://github.com/shapely/shapely/pull/2214

Closes #325