sea-bass/pyrobosim #331: Missing QT dependency for visualization window

mlsdpk, 2025-02-15:
During a fresh installation of pyrobosim without ROS, I faced an issue where the visualization window could not be opened due to a missing QT dependency. I needed to install `libxcb-cursor-dev` since it is required from Qt 6.5.0 onwards. I'm not sure if others face this problem, but I'm highlighting it here in case someone encounters the same issue.

OS: Ubuntu 24.04

sea-bass, 2025-02-19:
Interesting... Do you think it's worth adding a Troubleshooting page to the docs? Maybe issues and solutions like these can be collected there.

mlsdpk, 2025-02-23:
That would definitely be helpful. I'm not sure if there are other issues people are facing, but if there are more, it could be worth adding a section.