sea-bass/pyrobosim #336: Robot is not at an object spawn. Cannot detect objects.

PAyush15, 2025-03-10:
I am trying to create an action plan and execute it. It has mainly 4 steps: 1. navigate to location 2. detect objects 3. Pick the object 4. Place it back. Below is the code of the planner function:

```
def start_planner(world, args):
    """Define and execute an action plan"""

    # Wait for the GUI to load
    while not world.has_gui:
        time.sleep(1.0)
    time.sleep(1.0)

    robot1 = world.robots[1]
    objecttocheck = world.get_object_by_name("socket1")

    actions = [
        TaskAction("navigate", target_location="box2_box"),
        TaskAction("detect", robot=robot1),
        TaskAction("pick", robot=robot1, object="socket1"),
        TaskAction("place", object="socket1"),
    ]

    plan = TaskPlan(actions=actions)
    result, num_completed = robot1.execute_plan(plan)
    print("Details of the object parent: ", objecttocheck.print_details())
    print("Object spawn locations: ", world.get_object_spawns())
    print(result)
```

The navigation task works fine. But when I try to detect the object, it gives me an error. Below is the complete terminal output for the above func: 

_[world] INFO: Created world.
[IDOG-1] INFO: Created robot.
[IDOG-2] INFO: Created robot.
[world] INFO: Initialized PyRoboSim GUI.
[IDOG-2] INFO: Executing task plan...
[IDOG-2] INFO: Executing action navigate [1/2]
[IDOG-2] INFO: Action completed with result: SUCCESS
[IDOG-2] INFO: Executing action detect [2/2]
[IDOG-2] WARNING: Robot is not at an object spawn. Cannot detect objects.
[IDOG-2] INFO: Action completed with result: PRECONDITION_FAILURE
[IDOG-2] INFO: Task plan failed to execute on action 2/2
[IDOG-2] INFO: Task plan completed with status: PRECONDITION_FAILURE
Object: socket1 in box2_box
	Pose: [x=7.14, y=14.84, z=0.20, qw=0.930, qx=0.000, qy=-0.000, qz=0.368]
None
Object spawn locations:  [Object spawn: box1_box, Object spawn: box2_box, Object spawn: box3_box, Object spawn: box4_box]
Execution result with status: PRECONDITION_FAILURE_

I command the robot to move to "box2_box" which is a object spawn location (as seen in the output of terminal). But after it reaches there, it throws an error that it is not at an object spawn location. I checked the examples and tried different target_locations like "box2", "box2_box", "socket1" (which is the object at location "box2") but none works.

When I try to comment and pick the object "socket1" at location "box2", I get the following error: 

_Exception in thread Thread-1 (start_planner):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/nitro/ros2_ws/src/pyrobosim/pyrobosim/examples/idac_world.py", line 589, in start_planner
    result, num_completed = robot1.execute_plan(plan)
  File "/home/nitro/ros2_ws/build/pyrobosim/pyrobosim/core/robot.py", line 1006, in execute_plan
    result = self.execute_action(act_msg)
  File "/home/nitro/ros2_ws/build/pyrobosim/pyrobosim/core/robot.py", line 915, in execute_action
    result = self.pick_object(action.object, action.pose)
  File "/home/nitro/ros2_ws/build/pyrobosim/pyrobosim/core/robot.py", line 508, in pick_object
    message = f"{obj.name} is at {obj.parent.name} and robot is at {loc.name}. Cannot pick."
AttributeError: 'NoneType' object has no attribute 'name'_

Did someone face a similar error? Below is the full code to run my world and the thread for action plan:

sea-bass, 2025-03-11:
Happy to look into this -- what do your metadata files describing the `box` location and the `rfid`/`socket` objects look like?

sea-bass, 2025-03-11:
Actually, I just made up the data.

The reason for the failure is that 

```

sea-bass, 2025-03-11:
Actually, I just made up the data and it seems to work on main. Can you update your branch to `main` and try again?

You will need to make 2 small adjustments:

1. The imports here

```python
# from pyrobosim.navigation import (
#    ConstantVelocityExecutor,
#    AStarPlanner,
#    PRMPlanner,
#    RRTPlanner,
# )

from pyrobosim.navigation.execution import ConstantVelocityExecutor
from pyrobosim.navigation.rrt import RRTPlanner
```

2. The GUI checking here:

```python
#while not world.has_gui: 

while world.gui is None:
```

[image]

sea-bass, 2025-03-11:
Hmm, I even went back a few commits and your example still seems to work fine on my machine... maybe there's something about the location and/or object metadata you are using that we can check?

I basically just copied the `desk` (and made it smaller) for the `box`, and used `apple` and `banana` objects instead of `socket` and `rfid`.

PAyush15, 2025-03-11:
I checked with the desk as location, it works for me too! 

I think that there might be something in the way I define the box. However, I checked it and looked fine to me. Not sure there.

`box:
  footprint:
    type: polygon
    coords:
      - [-0.2, -0.2]
      - [0.2, -0.2]
      - [0.2, 0.2]
      - [-0.2, 0.2]
    height: 0.2
  locations:
    - name: "box"
      footprint:
        type: parent
      nav_poses:
        - position:
            x: 0.2
            y: 0.0
          rotation_eul:
            yaw: 1.57`

But for now, it isn't as relevant, so I can go ahead with the desk as the location. Thanks for the help!

closed 2025-03-11 by PAyush15

sea-bass, 2025-03-11:
Hmm, it might be that the nav pose defined above is in collision with the box itself? Try moving them out to like 0.4 m instead of 0.2. Basically accounting for the robot/inflation radius.