sea-bass/pyrobosim #359: Build failing as GPG key expired

SGeetansh, 2025-06-02:
During the Docker build for the `pyrobosim_ros` base image, the step

```sh
RUN apt-get update && apt-get upgrade -y
```

fails with the following error:

```sh
W: GPG error: http://packages.ros.org/ros2/ubuntu jammy InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
E: The repository 'http://packages.ros.org/ros2/ubuntu jammy InRelease' is not signed.
```

Tried: (from https://robotics.stackexchange.com/questions/99143/apt-update-signatures-were-invalid-f42ed6fbab17c654)
```sh
sudo curl -sSL https://raw.githubusercontent.com/ros/rosdistro/master/ros.key -o /usr/share/keyrings/ros-archive-keyring.gpg
```

But cURL is not installed. Can't install it because it needs indexes to be updated via ```apt update```. A little confused on what should be the long term solution here.

sea-bass, 2025-06-02:
Thanks for pointing this out. I believe this has to be done now:

https://discourse.ros.org/t/ros-signing-key-migration-guide/43937/8

So I think now just installing `ros2-apt-source` should do?

I can look into this soon too, I guess the repo has not run into it due to using old pushed images?

sea-bass, 2025-06-02:
Wait, this shouldn't be necessary since we're relying on the official ros images here...

https://github.com/sea-bass/pyrobosim/blob/22123f9ec22324f0eb9b5d7097f87ba33fd83f16/docker/Dockerfile#L54

Maybe those images just need a bit to be updated?

SGeetansh, 2025-06-02:
I see. I believe the same, just have to wait it out. Thanks for your attention to this matter!

sea-bass, 2025-06-03:
@SGeetansh should be good now!